rokath/trice #528: Does Trice support sending traces over UDP/TCP from MCU

srgg, 2025-05-14:
I'm implementing Trice on an ESP32-S3 board and would like to send traces wirelessly over WiFi to a PC using either UDP or TCP.

In the documentation, I see support for various output methods like UART and RTT, but I couldn't find clear information about network protocols. I've set up Trice on my ESP32-S3 and have implemented a WiFi connection, but I'm not sure about the best way to send the trace data over the network.

I've successfully implemented Trice on my ESP32-S3 board with UDP transmission for the trace data. The ESP32-S3 is properly connected to WiFi and is sending Trice packets via UDP to my PC's IP address and port.

However, I can't figure out how to receive and decode these traces on the PC side.

srgg, 2025-05-15:
Answering myself: UDP receiver wasn’t implemented — started working on it. Getting some packets from MCU, but clearly incomplete:

```
Encoding is TREX
inconsistent TCOBSv1 buffer!
sigil chain mismatch inconsistent TCOBSv1 buffer:
sigil chain mismatch 00000000  75 9b 75 9b e3 01 12 02  34 12                    |u.u.....4.|

inconsistent TCOBSv1 buffer!
sigil chain mismatch inconsistent TCOBSv1 buffer:
sigil chain mismatch 00000000  20 9c 20 9c ee 01 13 04  78 56 34 12 21 9c 21 9c  | . .....xV4.!.!.|
00000010  f7 01 14 08 f0 de bc 9a  78 56 34 12 e6 8a e6 8a  |........xV4.....|
00000020  ae 03 15                                          |...|

inconsistent TCOBSv1 buffer!
sigil chain mismatch inconsistent TCOBSv1 buffer:
sigil chain mismatch 00000000  b7 9d b7 9d 98                                    |.....|

May 14 23:24:35.735223  UDP4:                           0              ERROR:triceType typeX0 not implemented (hint: IDBits value?)
May 14 23:24:35.735312  UDP4:                           0              ERROR:triceType typeX0 not implemented (hint: IDBits value?)
Inconsistent data, discarding last single byte 22 from:
00000000  16                                                |.|

```

srgg, 2025-05-16:
A pull request has been opened to implement this feature #529.  It adds IPv4 UDP support to enable receiving trace data from an MCU.

rokath, 2025-05-16:
Trice in the current version should be able to receive over TCP: https://github.com/rokath/trice/blob/master/docs/TriceUserManual.md#1327-tcp-input

**Thanks a lot for the pull request.** I am able to have a closer look only in June.

srgg, 2025-05-16:
@rokath  Thank you for the fantastic tool—really appreciate your work! June is fine (I'm using a self-built version for now).

I saw the example, but it looked like it was capturing from a COM port and forwarding to TCP. In my case, I'm just using a UDP receiver with no wired connection. Since UDP doesn't require a persistent connection, it's a good fit for telemetry or debug in the field.

For reference, here's the example I saw:
```shell
trice l -p COM3 -tcp 127.0.0.1:23
```

rokath, 2025-05-16:
Yes, you are right. Trice can transmit over TCP and also receive over TCP. But UDP is a good enrichtment and I am looking forward to integrate your work.

rokath, 2025-05-23:
Hello sgg!

Thanks a lot for your very useful Trice add-ons. I really like them. I still have limited computer access. What I can tell so far: 

Some of the tests are failing now. Please run the testAll.sh script. Already `go test ./...` is failing. The failures are easy to fix, is my first assumption.

It would be very kind of you, if you write a few lines of documentation concerning your UDP4 extension and the alias switch, maybe with example usage included for both extensions.

It would be fine, if you just put a few lines in our conversation here. I will integrate them later into the Trice user manual.

srgg, 2025-05-25:
@rokath, please let me know if the PR lacks sufficient information about the changes I've made and usage examples.

 I have fixed the tests and updated the PR, please see test logs https://github.com/rokath/trice/pull/529#issuecomment-2907620914

Thank you for your time

PR #529 (2025-05-25): Add IPv4 UDP Receiver and -exclude Flag for Source Filtering
This pull request introduces two key enhancements:

1. **IPv4 UDP Receiver**  
   Adds support for receiving data over IPv4 using UDP. This enables integration with systems that broadcast or transmit telemetry, logs, or other messages over the network.

2. **`-exclude` Flag**  
   Introduces a command-line flag `-exclude` that allows users to specify one or more source addresses to be omitted from scanning or processing. This improves flexibility in environments with known noisy or irrelevant sources.

### `-port UDP4` Example

To re [... cut]
closed 2025-05-25 by rokath