rokath/trice #534: Log not working

fmlika, 2025-05-28:
So I am trying to test trice on my project. I wanted to see logs with an USB cable using trice l -p COM7 -baud 115200 -v . 
But I've got nothing in the terminal after that command. What am I doing wrong ? 
This is my triceConfig.h : 
`#ifndef TRICE_CONFIG_H_
#define TRICE_CONFIG_H_

#ifdef __cplusplus
extern "C" {
#endif

//! TRICE_CLEAN, if found inside triceConfig.h, is modified by the Trice tool to silent editor warnings in the cleaned state.
#define TRICE_CLEAN 0 // Do not define this at an other place! But you can delete this here.

// hardware specific trice lib settings
#include "main.h"
//#define TriceStamp16 TIM17->CNT     // 0...999 us
#define TriceStamp32 HAL_GetTick()  // 0...2^32-1 ms (wraps after 49.7 days)

#define TRICE_BUFFER TRICE_STATIC_BUFFER
//#define TRICE_DEFERRED_BUFFER_SIZE 2000

// Windows: trice log -p jlink -args "-Device STM32F030R8" -pf none -prefix off -hs off -d16 -showID "deb:%5d" -i ../../demoTIL.json -li ../../demoLI.json
// Unix:   ./RTTLogUnix.sh or manually:
// 		Terminal 1: rm -f ./temp/trice.bin && JLinkRTTLogger -Device STM32G0B1RE -If SWD -Speed 4000 -RTTChannel 0 ./temp/trice.bin
//      Terminal 2: touch ./temp/trice.bin && trice log -p FILE -args ./temp/trice.bin -pf none -prefix off -hs off -d16 -ts ms -i ../../demoTIL.json -li ../../demoLI.json
#define TRICE_DIRECT_OUTPUT 1
#define TRICE_USB_ENABLE 1

#define TRICE_DIRECT_SEGGER_RTT_32BIT_WRITE 1
#define BUFFER_SIZE_UP (2048) // "TRICE_DIRECT_BUFFER_SIZE"

#define TRICE_BINARY_ENABLE 1

// Windows: trice log -p com3         -prefix off -hs off -pw MySecret -pf COBS -i ../../demoTIL.json -li ../../demoLI.json
// Unix:    trice log -p /dev/ttyACM0 -prefix off -hs off -pw MySecret -pf COBS -i ../../demoTIL.json -li ../../demoLI.json

#include "cmsis_gcc.h"
#define TRICE_ENTER_CRITICAL_SECTION { uint32_t primaskstate = __get_PRIMASK(); __disable_irq(); {
#define TRICE_LEAVE_CRITICAL_SECTION } __set_PRIMASK(primaskstate); }

//void TriceHeadLine(char* name);
//void LogTriceConfiguration(void);
//void SomeExampleTrices(int burstCount);

#define TRICE_FULL_CHECK

#ifdef __cplusplus
}
#endif

#endif /* TRICE_CONFIG_H_ */`

rokath, 2025-05-28:
You configured RTT output but you are listen on a serial port. That cannot work.

For UART out, usuallally TRICE_DEFERRED needs to be enabled and you have to configure which UART you are going to use. First try to get a character over the wire to see if the communication works. See the examples.

To listen over RTT, as configured use adapted to your system

```C
// Windows: trice log -p jlink -args "-Device STM32F030R8" -pf none -prefix off -hs off -d16 -showID "deb:%5d" -i ../../demoTIL.json -li ../../demoLI.json
// Unix: ./RTTLogUnix.sh or manually:
// Terminal 1: rm -f ./temp/trice.bin && JLinkRTTLogger -Device STM32G0B1RE -If SWD -Speed 4000 -RTTChannel 0 ./temp/trice.bin
// Terminal 2: touch ./temp/trice.bin && trice log -p FILE -args ./temp/trice.bin -pf none -prefix off -hs off -d16 -ts ms -i ../../de
```

fmlika, 2025-05-30:
Update : using `$ C:/Users/userName/go/src/gostlink-master/stRttLogger.exe -if SWD "-RTTSearchRanges=0x20000000 0xA0000" -RTTChannel 0 -Speed 4000 >./temp/log.bin`
 in a terminal
and 
`trice log -d16 -p FILE -args=./temp/log.bin -i til.json -li li.json  -pf none -d16 -ts ms` 
In another solved the problem