rokath/trice #535: Cycle error

fmlika, 2025-05-30:
I am Getting this error : 

[image]

My TriceConfig.h is : 
/*! \file triceConfig.h
\author Thomas.Hoehenleitner [at] seerose.net
*******************************************************************************/

#ifndef TRICE_CONFIG_H_
#define TRICE_CONFIG_H_

#ifdef __cplusplus
extern "C" {
#endif

//! TRICE_CLEAN, if found inside triceConfig.h, is modified by the Trice tool to silent editor warnings in the cleaned state.
#define TRICE_CLEAN 0 // Do not define this at an other place! But you can delete this here.

//! ms32 is a 32-bit millisecond counter, counting circular in steps of 1 every ms.
//extern uint32_t ms32;
//#define TriceStamp16 (SysTick->VAL) // Counts from 31999 -> 0 in each ms.
//#define TriceStamp32 ms32
#include "main.h"
#define TriceStamp16 TIM17->CNT    // 0...999 us
#define TriceStamp32 HAL_GetTick() // 0...2^32-1 ms (wraps after 49.7 days)

#define TRICE_BUFFER TRICE_STATIC_BUFFER
#define TRICE_DEFERRED_BUFFER_SIZE 2000

// Windows: trice log -p jlink -args "-Device STM32F030R8" -pf none -prefix off -hs off -d16 -showID "deb:%5d" -i ../../demoTIL.json -li ../../demoLI.json
// Unix:   ./RTTLogUnix.sh or manually:
// 		Terminal 1: rm -f ./temp/trice.bin && JLinkRTTLogger -Device STM32G0B1RE -If SWD -Speed 4000 -RTTChannel 0 ./temp/trice.bin
//      Terminal 2: touch ./temp/trice.bin && trice log -p FILE -args ./temp/trice.bin -pf none -prefix off -hs off -d16 -ts ms -i ../../demoTIL.json -li ../../demoLI.json
#define TRICE_DIRECT_OUTPUT 1
#define TRICE_DIRECT_SEGGER_RTT_32BIT_WRITE 1
#define BUFFER_SIZE_UP (2048) // "TRICE_DIRECT_BUFFER_SIZE"

// Windows: trice log -p com3         -prefix off -hs off -pw MySecret -pf COBS -i ../../demoTIL.json -li ../../demoLI.json
// Unix:    trice log -p /dev/ttyACM0 -prefix off -hs off -pw MySecret -pf COBS -i ../../demoTIL.json -li ../../demoLI.json
#define TRICE_DEFERRED_OUTPUT 0

#define TRICE_CYCLE_COUNTER 1

#define TRICE_PROTECT 1

#define TRICE_DEFERRED_XTEA_ENCRYPT 1
#define TRICE_DEFERRED_OUT_FRAMING TRICE_FRAMING_COBS
#define TRICE_DEFERRED_UARTA 1
#define TRICE_UARTA USART2

#include "cmsis_gcc.h"
#define TRICE_ENTER_CRITICAL_SECTION             \
	{                                            \
		uint32_t primaskstate = __get_PRIMASK(); \
		__disable_irq();                         \
		{
#define TRICE_LEAVE_CRITICAL_SECTION \
	}                                \
	__set_PRIMASK(primaskstate);     \
	}

void TriceHeadLine(char const* name);
void LogTriceConfiguration(void);
void SomeExampleTrices(int burstCount);

#define TRICE_FULL_CHECK

#ifdef __cplusplus
}
#endif

#endif /* TRICE_CONFIG_H_ */

And the line of code is  : `trice(iD(2546), "Hello from STM32!\n");`

What can be the problem ?

rokath, 2025-05-30:
If you use RTT only, remove ALL lines containing any `...DEFERRED...` from your triceConfig.h. 

BUFFER_SIZE_UP should be set in SEGGER_RTT_Conf.h. Get yoown copy.

Just have a single `trice("hi\n");` at start up. Use the `-s -debug` switches to see what is ging on.

fmlika, 2025-05-30:
also, this is the log file content, with messae "hi" : 
[00:23:04]  INFO Welcome to goST-Link library rtt logger...
[00:23:04]  INFO Found st-link witch matching product and vendor id [374f, 0483]
[00:23:04]  WARN error occured while trying to leave mode: cannot leave mass storage mode
[00:23:04]  INFO got id code: 0be12477
[00:23:04]  INFO searching for SeggerRTT in range  [20000000, 200a0000]
[00:23:07]  INFO found RTT control block at address: 0x2008aaf8
[100 87 192 0]

rokath, 2025-05-30:
Your case is similar to configuration 88 in L432 example:
```C
#define TRICE_DIRECT_OUTPUT 1
#define TRICE_DIRECT_SEGGER_RTT_32BIT_WRITE 1
#define TRICE_BUFFER TRICE_STATIC_BUFFER
#define TRICE_PROTECT 1
#define TRICE_DIAGNOSTICS 1
```
If you have just one single trice and not repeated, you should get exactly 4 bytes. 

It looks, like you get an ASCII message.

fmlika, 2025-06-02:
Ok so this is the output with `trice log -d16 -p FILE -args=./temp/log.bin -i til.json -li li.json  -pf none -d16 -ts ms -v
`
: 

[image]

I think it is trying to decode the log.bin file content which is text !

fmlika, 2025-06-02:
Update : it seems it is the way I wrote my logging command that causes problem, with this command it works : 

[image]