rokath/trice #545: Symbols such as `<` gets toggled between Unicode and symbol representation

dbeliavskij, 2025-07-01:
**Describe the bug**
Having some trices in code with symbols `<` inside a format string, I noticed that these randomly switch between unicode representation and actual symbol representation:

[image]

**To Reproduce**
Add trices with `<` symbol, execute `trice insert`, perform other changes in code executing `trice insert` again until symbols change. I could not detect what exactly causes a switch and it does not happen every time.

**Expected behavior**
Trices do not change without change to trice itself.

dbeliavskij, 2025-07-01:
The file encoding is UTF-8

I am not sure how the editor can be related. What I shoved in the screenshot is what change git sees in `til.json` file. The trices themselves in the code do not change, only their representation in `til.json`

rokath, 2025-07-01:
Currently no idea. For the conversion and file IO library functions are used:

```Go

// toJSON converts lut into JSON byte slice in human-readable form.
func (lu TriceIDLookUp) toJSON() ([]byte, error) {
	return json.MarshalIndent(lu, "", "\t")
}

// toFile writes lut into file fn as indented JSON and in verbose mode helpers for third party.
func (ilu TriceIDLookUp) toFile(fSys afero.Fs, fn string) (err error) {
	var fJSON afero.File
	fJSON, err = fSys.Create(fn)
	msg.FatalOnErr(err)
	defer func() {
		err = fJSON.Close()
		msg.FatalOnErr(err)
	}()
	var b []byte
	b, err = ilu.toJSON()
	msg.FatalOnErr(err)
	_, err = fJSON.Write(b)
	msg.FatalOnErr(err)
	/////////
	return
}
```

rokath, 2025-12-12:
There are now 2 tests in [id/utf8_test.go](https://github.com/rokath/trice/blob/main/internal/id/utf8_test.go). Closing this issue now. If the problem pops up again, please re-open, we can try to intensify those tests then to be able to catch and debug it.

closed 2025-12-12 by rokath